tierneytim/btAudio #2: Bluetooth information

Sixonlon, 2020-12-19:
Hi man, 

Thks for your wonderfull job !

 I wonder if there would be a way to retrive information from the music that is streamed to the ESP32? I think of title, album, etc

tierneytim, 2020-12-19:
Thanks man, 

I think so using the [AVRC](https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/components/bt/host/bluedroid/api/include/api/esp_avrc_api.h#L158)  profile but it may be complicated for two reasons. 1) the example code to use the profile is [hard to understand](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main) and 2) it requires that the device sending the audio also uses this profile which there is no guarantee of. 

What is the application you are thinking of?

Sixonlon, 2020-12-19:
I'm trying to convert an old Phillips car radio to a modern system. I wire up a DAB+ shield to my ESP32 and add the blutooth function using your work but i'd like to display music info on the LCD. I can do it with the DAB shield for the radio part but I struggle with bluetooth one

tierneytim, 2020-12-20:
Sounds Like a cool idea. This will probably take some time to get working but I think it will be a very useful addition. Will close issue when implemented.

tierneytim, 2020-12-25:
OK, metadata queries are now possible with the updateMeta method. Examples are provided for [printing, title, artist, album and genre](https://github.com/tierneytim/btAudio/blob/master/examples/queryMeta/queryMeta.ino) to the serial port. Let me know if this works for you.

Sixonlon, 2020-12-25:
Thks for the responsivness. The qeries work perfectly. Thks for your work and merry christmass

closed 2020-12-25 by Sixonlon

haldi4803, 2021-04-03:
If one would like to expand the metadata with other info...
The 	void updateMeta(); asks in the btaudio.cpp
`void btAudio::updateMeta() {
  uint8_t attr_mask = ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_ARTIST | ESP_AVRC_MD_ATTR_ALBUM | ESP_AVRC_MD_ATTR_GENRE;
  esp_avrc_ct_send_metadata_cmd(1, attr_mask);
}`
but the whole definition happens in the next function.
`void btAudio::avrc_callback(esp_avrc_ct_cb_event_t event, esp_avrc_ct_cb_param_t *param) {
  esp_avrc_ct_cb_param_t *rc = (esp_avrc_ct_cb_param_t *)(param);
  char *attr_text;
  String mystr;

  switch (event) {
    case ESP_AVRC_CT_METADATA_RSP_EVT: {
      attr_text = (char *) malloc (rc->meta_rsp.attr_length + 1);
      memcpy(attr_text, rc->meta_rsp.attr_text, rc->meta_rsp.attr_length);
      attr_text[rc->meta_rsp.attr_length] = 0;
      mystr = String(attr_text);

      switch (rc->meta_rsp.attr_id) {
        case ESP_AVRC_MD_ATTR_TITLE:
          //Serial.print("Title: ");
          //Serial.println(mystr);
		  title= mystr;
          break;
        case ESP_AVRC_MD_ATTR_ARTIST:
          //Serial.print("Artist: ");
          //Serial.println(mystr);
          artist= mystr;
		  break;
        case ESP_AVRC_MD_ATTR_ALBUM:
          //Serial.print("Album: ");
          //Serial.println(mystr);
          album= mystr;
		  break;
        case ESP_AVRC_MD_ATTR_GENRE:
          //Serial.print("Genre: ");
          //Serial.println(mystr);
          genre= mystr;
		  break;
      }
      free(attr_text);
  }break;
    default:
      ESP_LOGE(BT_RC_CT_TAG, "unhandled AVRC event: %d", event);
      break;
  }
}`

Seeing as up to 15 commands can be used. How would one go to include ESP_AVRC_MD_ATTR_PLAYING_TIME to get an audio.playingtime value to create a progess bar?
Sadly i'm not very well versed in c++ or programming in general.