tierneytim/btAudio #4: Sound is pitched down

BananaLoaf, 2021-01-20:
Sound is pitched down, songs (played via Spotify) start slowed down but later (4-5 sec) get to their normal speed.
```C
#include <btAudio.h>

btAudio audio = btAudio("BTSink");

void setup() {
  audio.begin();
 
  int bck = 26; 
  int ws = 22;
  int dout = 25;
  audio.I2S(bck, dout, ws);
}

void loop() {}
```
I use Wemos D1 mini ESP32 + PCM5102 I2S DAC Decoder Board
[image]

I'm aware that problem might be specific to my hardware or software(pipewire), but it does not persist when I use https://github.com/pschatzmann/ESP32-A2DP or https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink

tierneytim, 2021-01-20:
interesting, have never come across this before and don't have the same hardware so I will struggle to reproduce. To help me isolate the problem could you do the following

1. Run the code in the [underthehood](https://github.com/tierneytim/btAudio/blob/master/examples/underTheHood/underTheHood.ino) example and see if the problem persists. This essentially is a shorter version of a2dp sink example. 
2. put a delay in the void loop.
3. reduce the size of [.dma_buf_count and .dma_buf_len](https://github.com/tierneytim/btAudio/blob/5a2df6e082da3f061b7b9d72c19eb78c788dd9c9/examples/underTheHood/underTheHood.ino#L55).

BananaLoaf, 2021-01-20:
1. No effect
2. No effect
3. Reducing nor increasing .dma_buf_count and .dma_buf_len. did not help. However, I was able to sort of reproduce the effect while playing from phone with
```C
.dma_buf_count = 2,
.dma_buf_len = 8,
```

Even changing [.communication_format](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/main.c#L64) did not help

tierneytim, 2021-01-21:
very puzzling, 

some more questions.
1. Is this the only code you are running on your ESP32? 
2.  is this effect only present when using one device/piece of software or does it occur for every Bluetooth device you have? 
3. Could you say roughly by what factor the audio slows down? 
4. Does it last 4-5 seconds on every song or just on startup?

BananaLoaf, 2021-01-21:
1. Yes. Only BT Sink yet
2. It is present only on my Linux Laptop, affects every piece of software. Phone audio is fine.
3. About 1.32x
4. Every song
5. Even though I specify .sample_rate = 44100, when I connect to it, logs say it's 48000
```
E (23042) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (23042) BT_AV: A2DP connection state: Connecting, [04:ea:56:3e:a4:df]
I (23062) BT_AV: A2DP audio stream configuration, codec type 0
I (23062) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 16, CLKM: 13, BCK: 8, MCLK: 12292917.167, SCLK: 1538432.000000, diva: 64, divb: 1
I (23072) BT_AV: Configure audio player 11-15-2-33
I (23072) BT_AV: Audio player configured, sample rate=48000
```
Yet when I connect my phone it is set to 44100
```
E (274662) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (274672) BT_AV: A2DP connection state: Connecting, [f0:5c:77:d9:85:93]
I (274792) BT_AV: A2DP audio stream configuration, codec type 0
I (274792) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11
I (274802) BT_AV: Configure audio player 21-15-2-35
I (274812) BT_AV: Audio player configured, sample rate=44100
```
Sample rate does not change while playing audio

Manually changing sample rate to 48000 fixed pitch and slowdowns **on the PC**, but phone audio is now **_pitched up_**

tierneytim, 2021-01-21:
Great work. I think you're laptop is requesting 48000 and this isn't being acknowledged properly in my code but is in the IDF code.  I think I just need to write in a callback to flexibly adapt to different sampling rates. Will work on this over the next few days.

BananaLoaf, 2021-01-21:
Thank you for your work! I really appreciate it!

tierneytim, 2021-01-21:
will forget if closed

tierneytim, 2021-01-22:
OK, fix to determine sample rate automatically is live. I can't really test this for any 48KHz source so if you could test and report back that would be great.

BananaLoaf, 2021-01-23:
Unfortunately, bug is still present

tierneytim, 2021-01-23:
With the debug notifications turned on what is the output?

tierneytim, 2021-01-23:
OK, attempt 2 is up and should change the sample rate. the previous attempt only worked if the connection was made before i2s configuration.

BananaLoaf, 2021-01-23:
The output for phone audio was
```
[I][btAudio.cpp:77] a2d_cb(): A2DP audio stream configuration, codec type 0
[I][btAudio.cpp:93] a2d_cb(): Configure audio player 21-15-2-35
[I][btAudio.cpp:94] a2d_cb(): Audio player configured, sample rate=44100
[E][btAudio.cpp:144] avrc_callback(): unhandled AVRC event: 0
[E][btAudio.cpp:144] avrc_callback(): unhandled AVRC event: 5
[E][btAudio.cpp:144] avrc_callback(): unhandled AVRC event: 0
```
And for PC
```
[I][btAudio.cpp:77] a2d_cb(): A2DP audio stream configuration, codec type 0
[I][btAudio.cpp:93] a2d_cb(): Configure audio player 11-15-2-33
[I][btAudio.cpp:94] a2d_cb(): Audio player configured, sample rate=48000
[E][btAudio.cpp:144] avrc_callback(): unhandled AVRC event: 0
[E][btAudio.cpp:144] avrc_callback(): unhandled AVRC event: 5
[E][btAudio.cpp:99] a2d_cb(): a2dp invalid cb event: 1
[E][btAudio.cpp:144] avrc_callback(): unhandled AVRC event: 0
[E][btAudio.cpp:99] a2d_cb(): a2dp invalid cb event: 1
```

BananaLoaf, 2021-01-23:
But now at all seems to be working correctly, although I'm getting a little bit of static noise, but I guess that's just my crappy speakers or cable

tierneytim, 2021-01-25:
Is this static specific to my code or do you observe it with the IDF code as well? I sometimes get bad "static" from bad wiring, amplifier not having enough power or the gain on the amplifier (max98357A) changing due to interference.

BananaLoaf, 2021-01-29:
I'm not much of the expert in this, but static went away after I disconnected DAC board from ground (???). I have not checked other examples for static with GND plugged in, but I will let you know if I discover something.

tierneytim, 2021-02-15:
closing as initial pitch issue solved

closed 2021-02-15 by tierneytim