tierneytim/btAudio #7: Sending AVRCP commands

lmore377, 2021-03-22:
I figured out how to send some AVRCP commands. Basically you just run this, replacing `<command>` with one of the commands below.

`esp_avrc_ct_send_passthrough_cmd(0, <command>, ESP_AVRC_PT_CMD_STATE_PRESSED); `

You can see all the commands [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html#_CPPv417esp_avrc_pt_cmd_t) but it seems like only the ones below work (I tried just the hex codes for other commands but my phone didn't respond to any except these. I don't know if they're not supported by the avrc library or my phone).
```
ESP_AVRC_PT_CMD_PLAY
ESP_AVRC_PT_CMD_STOP
ESP_AVRC_PT_CMD_PAUSE
ESP_AVRC_PT_CMD_FORWARD
ESP_AVRC_PT_CMD_BACKWARD
ESP_AVRC_PT_CMD_REWIND
ESP_AVRC_PT_CMD_FAST_FORWARD
```
You can see more info about that specific method [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html#_CPPv432esp_avrc_ct_send_passthrough_cmd7uint8_t7uint8_t7uint8_t).

tierneytim, 2021-03-24:
Great Idea to get these commands built-in. I've been snowed under by work recently But I can see the light again and will add this to the to-do list.

biglee991228, 2021-03-25:
I tested but only "play","stop"  and  "pause"  functions worked.
here's my loop()
if(Serial.available()){
  //read until a terminator. after this point there should only be numbers
  command = Serial.readStringUntil('#');
  if(command.equals("play")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("play");
    }
  if(command.equals("pause")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("pause");
    }  
  if(command.equals("FF")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FAST_FORWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("fast_forward");
    }  
  if(command.equals("RW")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_REWIND, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("rewind");
    }  
  if(command.equals("stop")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("stop");
    }  
  if(command.equals("forward")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("forward");
    }
   if(command.equals("backward")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   Serial.println("backward");
    }
 }

lmore377, 2021-03-25:
@biglee991228 fast forward and rewind didn't work for me either but everything else does. Have you tried using a different app and/or device?

biglee991228, 2021-03-29:
@lmore377 
I got it!
if(Serial.available()){
  //read until a terminator. after this point there should only be numbers
  command = Serial.readStringUntil('#');
  if(command.equals("play")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("play");
    }
  if(command.equals("pause")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("pause");
    }  
  if(command.equals("stop")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_STOP, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("stop");
    }  
  if(command.equals("forward")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_FORWARD, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("forward");
    }
   if(command.equals("backward")){
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_PRESSED);
   esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_BACKWARD, ESP_AVRC_PT_CMD_STATE_RELEASED);
   Serial.println("backward");
    }
 }
don't forget "RELEASED“！

lmore377, 2021-03-29:
My device doesn't seem to need/respond to the released commands but I'll keep that in mind if something pops up

biglee991228, 2021-03-29:
my phone can be controlled by my chevey car (linux OS),so it must can  be controlled by the passthrough command  from ESP32.
because  the  AVRCP profile is the same thing.

haldi4803, 2021-04-03:
Great idea.
I have 3 buttons on my EPS32 board which seems like a perfect use case for that!

ES-KO, 2021-04-27:
Is it somehow possible to detect, if the device is connected?

tierneytim, 2021-05-16:
for checking connection events see [here ](https://github.com/tierneytim/btAudio/blob/dcd4bf5eb592377afc28112ffa10681e0a151716/src/btAudio.cpp#L65). I could add a variable to the class that monitors this connection.

ES-KO, 2021-05-17:
That would be pretty cool,
cause my programming skills are just on a basic level …

brad-colbert, 2021-07-05:
What about "receiving" volume control command?  Is that possible (assuming my streaming device sends them)?