tierneytim/btAudio #8: a2dp source

yeckel, 2021-05-15:
Hi Tim,
how complicated would be to add a2dp source feature? I would like to decode something on the ESP32 and send it to Bluetooth headset. (Have 8k 16bit signed mono data) I know that HFP_GW is not (yet) supported by the arduino_esp32 but a2dp source shall work.

tierneytim, 2021-05-16:
Hi Yeckel, 

This has been on the to-do list but like so many other things my real job gets in the way. I have periodically looked at the a2dp source example on the ESP32-IDF but my head starts to hurt.[ It's over 800 lines long](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source/main) and is, to my brain, very complicated. So... this is on to-do list but can't say when.

yeckel, 2021-05-16:
Yes I know, the same here. I don't want to port my app and libs used into ESP-IDF and arduino as a component in ESP-IDF doesn't work well.

closed 2021-05-16 by yeckel