tierneytim/btAudio #9: Request Record feature

rmond18, 2021-05-16:
Hi, I've been trying to develop a feature by adding a record capability where it is triggered by a push button to record data received to a connected microSD card. Unfortunately I'm quite confused using the library since I'm new to object-oriented programming. May I ask for help for a sample code for this if possible?

You have advised that:
> If you go this [line](https://github.com/tierneytim/btAudio/blob/dcd4bf5eb592377afc28112ffa10681e0a151716/src/btAudio.cpp#L250) you can add a RECORD case by simply writing to an SD card using SDIO.

but I'm quite lost how to do it.

You also said that:
> This isn't in the code because you could then easily stream copyrighted music from the internet to an SD card. That would be illegal...

I'll be using this receiver together with another ESP32 as a BT source connected to an i2s mic.

Thank you very much.
Warm regards,
rmond

tierneytim, 2021-05-16:
Hi rmond, this feature won't be put online because It could be abused by others. I'd start off learning the different SD interfaces for ESP32, if you don't already know them. It will need to be as fast as possible and the standard SPI interface may not be sufficient.  See this instructable for how to [benchmark](https://www.instructables.com/Select-SD-Interface-for-ESP32/)

Once you get a fast enough SD writer code for your application you can write a custom [callback](https://github.com/tierneytim/btAudio/blob/dcd4bf5eb592377afc28112ffa10681e0a151716/examples/customCallbacks/customCallbacks.ino#L12). The variable data16 has the audio data and you need to write this block of data to the SD card. 

hope this helps

rmond18, 2021-06-10:
Hi Tim,

Thank you for your help. I'm still struggling with the actual code for saving the audio data from variable data16 to SD card.

Currently, my reference for writing the data to the sd card is this [sound recorder code](https://github.com/MhageGH/esp32_SoundRecorder) by Mhage, however I'm having problems on how to modify the this code since in the sound recorder code, the data is coming from an I2S pinout for the data. While in the custom callback, the data is already in a variable since it came from a bt source.

Do you have any tips/sample for the coding of the actual writing of data to sd card?

closed 2021-07-01 by tierneytim