tierneytim/btAudio #12: MAX98357A

darkmoon3d, 2021-07-01:
Should this work on MAX98357A (https://www.adafruit.com/product/3006) or will this only work on I2S Stereo Decoder - UDA1334A (https://www.adafruit.com/product/3678)?

Could code be modified to work with PWM audio Pam8302-(https://www.adafruit.com/product/2130)?

tierneytim, 2021-07-01:
1. Yes, it will work with MAX98357A. I made a lipo-powered Bluetooth speaker with the [Sparkfun thing plus](https://shop.pimoroni.com/products/sparkfun-thing-plus-esp32-wroom?gclid=CjwKCAjwz_WGBhA1EiwAUAxIccNRdfRjGGVNgPkh20a3-zCsNEB1i2zE0hoeZgG27mdictYIaD7R5BoCtT8QAvD_BwE). Whatever board you use you need to make sure the power supply can supply as much current as the board needs. the thing plus has a beefy regulator and lipo management making it ideal. 
2. Could do but I'm currently working on PDM audio as opposed to PWM audio so I'm unlikely to build that in anytime soon.

closed 2021-07-01 by tierneytim

darkmoon3d, 2021-07-01:
Interesting. Am using similar setup with Huzzah ESP32 board (https://www.adafruit.com/product/3405) with MAX98357A but unable to work. ESP32 device 'ESP_Speaker' is showing up but not getting any audio. Using 3W 4ohm speaker (https://www.adafruit.com/product/4445)

Using following wiring:

ESP32 == MAX98357A
D25 == DIN
D26 == BCLK
F27 == LRC
3V == VIN
GND == GND

Have also tried USB==VIN

Arduino Settings:
Board: Adafruit ESP32 Feather
Upload Speed: 11520
Flash Freq: 80mhz
Partition: Default
Core Debug: None
Port: COM11

darkmoon3d, 2021-07-09:
Tested out using 3678 and managed to get connection and audio but getting noise only. Same setup as before aside that D27==WSEL.

darkmoon3d, 2021-07-09:
Swapped pins 25 and 26. Working now. Also plugged PAM8302 onto each stereo out channel.

tierneytim, 2021-07-09:
Brilliant, glad it's working now