tierneytim/btAudio #13: Crash with BT esp_bluedroid_init and WIFI

brad-colbert, 2021-07-05:
Awesome project!

How did you get around the crash when BT and WIFI are enabled?

I've tried two different boards (Adafruit HUZZAH, and DOIT ESP32 DEVKIT V1) with the same result.  System crashes when WIFI and BT are enabled.

Arduino 1.8.15.

tierneytim, 2021-07-05:
thanks, man, 

To clarify, Is it my code that crashes the system (the webserver), or is it that some other WIFI code (that someone else wrote) causes a crash when used with my library.  Could you provide an example that crashes your system?

brad-colbert, 2021-07-05:
Hi Tim,

No, it's not your code! Sorry!  I suspect it's some combination of the ESP32 Wroom (dual core) and Arduino.  Crash happens at the point that the WIFI is brought up (after BT is init'd) and goes away when either call is commented out (esp_bluedroid_init or WiFi.begin).  Just wondering if there was some magic incantation to get the WIFI and BT to play together.  Google is pretty sparse.

brad-colbert, 2021-07-05:
Found it!

I had to add the following to my _**~/Documents/Arduino/hardware/espressif/esp32/boards.txt**_ file:

`esp32doit-devkit-v1.menu.PartitionScheme.huge_app=Huge APP (3MB No OTA/1MB SPIFFS)
esp32doit-devkit-v1.menu.PartitionScheme.huge_app.build.partitions=huge_app
esp32doit-devkit-v1.menu.PartitionScheme.huge_app.upload.maximum_size=3145728
`

closed 2021-07-05 by brad-colbert