tierneytim/btAudio #14: Stream and record simultaneously

bluecaratmoa, 2021-07-05:
Is it possible to record the data received while listening to the audio simultaneously?

I'm planning to apply your code with a Bluetooth microphone as a source, I tried using [this](https://github.com/tierneytim/btAudio/blob/master/examples/customCallbacks/customCallbacks.ino), however, when I added the recording feature inside the callback function, the streaming stops. 

Is there other way to retrieve the bt data that wont disturb the streaming of bt data?

tierneytim, 2021-07-05:
It's really tough and depends how you want to record the data. If you want to use an sd card you'll need to guarantee you've got>>176.4 KB/s write speed. Then you need to allow for the i2s transmission as well. Not saying its impossible to do both but I haven't managed it.

bluecaratmoa, 2021-07-05:
I see. I have an SD card with minimum write speed of 10 MB/s. I'm not sure if I understood the callback function right, upon calling `audio.setSinkCallback(bt_data_cb1);` and then calling `audio.createFilter` will not affect the data anymore? I asked this because I'm planning to filter the data before saving it.

Also, this might be out of the scope of this repository but I would like to ask also if you are familiar on how can I write the bt data into an audio file?

tierneytim, 2021-10-04:
this feature will not be implmented

closed 2021-10-04 by tierneytim