tierneytim/btAudio #15: ESP Arduino Core 2.x breaks the library

easytarget, 2021-09-07:
ESP finally released their IDE core 2.0.0, breaking a couple of my projects in the process :-(
..and it seems to have broken this library; trying to compile the vanilla `advertiseBluetooth.ino` sketch with a fully updated IDE 1.8.15 results in:
```
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp: In member function 'void btAudio::begin()':
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:54:28: error: 'ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE' was not declared in this scope
   esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:54:28: note: suggested alternative: 'ESP_BT_GENERAL_DISCOVERABLE'
   esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                            ESP_BT_GENERAL_DISCOVERABLE
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp: In member function 'void btAudio::I2S(int, int, int)':
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:164:60: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
     .communication_format = static_cast<i2s_comm_format_t>(I2S_COMM_FORMAT_I2S|I2S_COMM_FORMAT_I2S_MSB),
                                                            ^~~~~~~~~~~~~~~~~~~
In file included from /home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/hal/esp32/include/hal/i2s_ll.h:30,
                 from /home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/hal/include/hal/i2s_hal.h:28,
                 from /home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/owen/Arduino/libraries/btAudio/src/btAudio.h:10,
                 from /home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:1:
/home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:70:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:164:80: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
     .communication_format = static_cast<i2s_comm_format_t>(I2S_COMM_FORMAT_I2S|I2S_COMM_FORMAT_I2S_MSB),
                                                                                ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/hal/esp32/include/hal/i2s_ll.h:30,
                 from /home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/hal/include/hal/i2s_hal.h:28,
                 from /home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/owen/Arduino/libraries/btAudio/src/btAudio.h:10,
                 from /home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:1:
/home/owen/.arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:71:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
/home/owen/Arduino/libraries/btAudio/src/btAudio.cpp:170:3: warning: narrowing conversion of 'btAudio::_sampleRate' from 'uint32_t' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
   };
   ^
exit status 1
Error compiling for board ESP32 Dev Module.
```
uurgh; a quick search shows that ESP have, indeed, messed with this recently, renaming some of the defines, eg [here](https://github.com/espressif/esp-idf/commit/5beb2802e0b6ba3b533ca08ac78c13aa26d1fc2d), and I assume elsewhere too.. I didn't look much further since.. priorities. But thought I should log this here.

In case you are interested, I'm using this on a esp32 powered watch, [code here](https://easytarget.org/ogit/owen/TTWatch-BluetoothAudio), based on an example in the TTGO watch library.

easytarget, 2021-09-07:
FYI, rolling back to ESP Core 1.0.6 in the IDE did the trick, and let me recompile, upload and use my sketch successfully.
PS: Thankyou for the code, I've used this as a late-night bedside Podcast player quite a lot during covid.

tierneytim, 2021-09-07:
Thanks man for picking this up. Glad you like it. So much for backwards compatibility on the ESP side. will hopefully get to this in the next week or so.

tierneytim, 2021-10-04:
all merged. good work on this

closed 2021-10-04 by tierneytim