tierneytim/btAudio #17: Bluetooth Connection Status

PilotAudio, 2021-09-23:
Hi!

Thank you so much for you work, it really has helped out with my project and works flawlessly.

I was just wondering if there was a way to perform some sort of Bluetooth connection status check?

I am currently working on a Bluetooth speaker and would love to be able to add a Bluetooth connection status light to show the user that they have successfully connected to the speaker but cannot figure out how to do it.

Any pointers in the right direction towards how to actually detect that a device has connected successfully would be greatly appreciated as I am fully stuck with this.

Thanks again!!

HiKE-yt, 2022-09-11:
@tierneytim @PilotAudio Hi,
I also need this to make a Bluetooth enabled HiFi system. Is there a way to do it ?
Thanks!

qwertychouskie, 2022-09-12:
Check the `btAudio::a2d_cb` function: https://github.com/tierneytim/btAudio/blob/master/src/btAudio.cpp#L100

It should be possible to add a bit of code there that does what you are looking for.