tierneytim/btAudio #20: Hands-free calling?

qwertychouskie, 2022-06-05:
My use case for this library is to upgrade my car's stereo system with Bluetooth.  What would be the best way to go about handling routing a microphone input back to the phone?

tierneytim, 2022-06-05:
The hands free protocol is not supported. But maybe there is a way around it with A2DP, depending on your application, 

What is the microphone in this case and why does it need to go to a phone? I'm not sure I understand what you are trying to do

qwertychouskie, 2022-06-05:
Maybe my wording is a bit weird.  Basically I just want to fully support hands-free calling (phone audio goes over car speakers, microphone is mounted in dash).  Probable will use https://www.adafruit.com/product/3421 for mic and https://www.mouser.com/ProductDetail/Adafruit/3678?qs=W0yvOO0ixfEeEgy1riJcig%3D%3D for audio output to car's audio channels.

More details here: https://www.reddit.com/r/arduino/comments/v591g8/looking_for_help_with_car_stereo_bluetooth_upgrade/

lmore377, 2022-06-05:
Looks like ESP-IDF supports both but as far as I know the Arduino library doesn't

https://github.com/espressif/esp-adf/tree/master/examples%2Fget-started%2Fpipeline_a2dp_sink_and_hfp

tierneytim, 2022-06-05:
both the mic and DAC are great, I use them all the time. However, i don't really know where to start with the hands-free stuff. I naively assumed the a2dp protocol would stream the phone audio(during a phone call) to my Bluetooth speaker and it didn't. Hopefully, the example from @lmore377  will be of more help

qwertychouskie, 2022-06-05:
Thanks for the pointers!  Quick question, does the ESP32-PICO-KIT (https://www.mouser.com/ProductDetail/356-ESP32-PICO-KIT) recommended in the README support multiple I2S devices?  I would hate to order everything just to find out that I can't hook up both the mic and audio output.

tierneytim, 2022-06-05:
yup, there are 2 I2S ports that support different sample rates between devices.

qwertychouskie, 2022-06-07:
Perhaps a little off topic to the code side of things, but any chance I could get a quick lookover of my Mouser parts list before I order to make sure I'm not missing anything/doing something stupid?  [Cart_Jun06_0945PM.pdf](https://github.com/tierneytim/btAudio/files/8849670/Cart_Jun06_0945PM.pdf)

This is the interfacing circuit: https://stuartschmitt.com/e_and_c_bus/interface.html

Thanks so much for all the help so far!

ericblade, 2022-07-20:
I may be wrong, but I think you're looking for the HSP profile, which is audio in/out, the HFP is used for dial/answer commands.   At least, my car that has built in bluetooth connectivity (but not A2DP, why???? sigh) connects to both HFP and HSP.

xRealNeon, 2022-11-02:
Is there anything new? As far as I can see this function only works with the esp framework and not with Arduino right?