tierneytim/btAudio #24: Error  audio.reconnect();

gt3rs40, 2022-07-26:
I have problems integrating the reconnect option

error message:
C:\Users\X\Documents\Arduino\sketch_jul26a\sketch_jul26a.ino: In function 'void setup()':
sketch_jul26a:12:8: error: 'class btAudio' has no member named 'reconnect'
  audio.reconnect();
        ^
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\X\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Nicht benutzt: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
'class btAudio' has no member named 'reconnect'

whats my mistake here?

thank you so much

tierneytim, 2022-07-26:
I have just merged the pull request that implements this method. download the updated library and report back. Let me know if this works

gt3rs40, 2022-07-26:
Thank you very much for your quick response. It worked well :) thanks again!

closed 2022-07-26 by tierneytim