tierneytim/btAudio #26: System latency of using Biquads

P-Mouse, 2022-10-18:
First of all, I am really liking this btAudio project!! 
Awesome work! :)

This is not an issue, but more like a question if you know how much the system latency is when using the IIR biquad filters?
I am purely talking about how much time the ESP32 needs to compute, not the external ADC/DAC.

Also, do you have any idea how many biquads can be used in total?

tierneytim, 2022-10-18:
Thanks for the kind words. No idea to be honest. I might have a look at writing a benchmark and putting it in the examples

P-Mouse, 2022-10-20:
> Thanks for the kind words. No idea to be honest. I might have a look at writing a benchmark and putting it in the examples

I would be extremely curious about this, so please keep me updated :)
Meanwhile I will keep digging, but I can't find any information on this.

I can also test this myself, I would only not know how to measure this?

On an higher level, you can probably measure some kind of signal or trigger before the IIR filter loop and afterwards I guess.

tierneytim, 2022-10-30:
ok, so I have uploaded my first attempt at [this](https://github.com/tierneytim/btAudio/blob/master/examples/biquadSpeedTest/biquadSpeedTest.ino). It would appear that the latency is about 300 ns/sample or 1.5% of CPU time per biquad per sample. So in theory, as an absolute maximum, you could use about 50 biquads....  but this depends on what else your application is doing. 

This number may be wrong as undoubtedly the benchmark can be improved but these results are consistent with my other projects that use 4th-order filters at 1.5 MHz for [sigma-delta modulation](https://github.com/tierneytim/Pico-USB-audio). Hope this helps. Closing now

closed 2022-10-30 by tierneytim