tierneytim/btAudio #30: Playing Bluetooth Audio through the built in DAC of the ESP

MrTwisterFister69, 2022-11-14:
Hey,
First of great work!
I was wondering how to output the Bluetooth audio instead of an I2S signal, as an analog audio signal by using the built in DAC of the ESP. I couldn´t find anything about it in your libary. 
I hope this is the right place to post my request, I am new to github.

Thanks in advance.

qwertychouskie, 2023-06-04:
I'm sure it's possible (there are videos on YouTube showing using the internal DAC, just not with this library), but the audio quality from the 8-bit (a.k.a. 0-255) DAC is going to result in extremely poor sound.  It is highly recommended to use an I2S DAC instead, as the audio will actually be of a usable quality.

daviti231, 2024-03-28:
any guidance to use internal dac in this library?

colonket, 2024-07-08:
I am also interested in finding examples of this. Any links to videos or webpages that explain this are greatly appreciated.

So far I have found these projects that require an amplifier board to make the sound loud enough to hear:
* https://docs.sunfounder.com/projects/esp32-starter-kit/en/latest/arduino/basic_projects/ar_bluetooth_audio_player.html 
* https://circuitdigest.com/microcontroller-projects/esp32-based-audio-player

but I have not been able to get them to work so far